PrintM3D/Crane #4: Crane-Quad-MENU

ericl-m3d, 2018-11-28:
Cannot access the s_prepare.

[image]

shoreliner, 2018-11-28:
Please explain when this issue occurs:

-On Startup?
-When Accessing a particular menu option?

ericl-m3d, 2018-11-28:
Clicked the prepare button at the start up. Then this came up

ericl-m3d, 2018-11-28:
This is occurring with the 2.02RC4 firmware version.

shoreliner, 2018-11-28:
Should only be using firmware version 2.02RC2 -- I have tested locally and menu is still functional.  While the latest firmware is 2.02RC5, we have not decided to move to it yet.

shoreliner, 2018-11-29:
Per our conversation we will begin testing for 2.02RC4; in the meantime this issue is resolved for the current version. Thanks!

shoreliner, 2018-11-29:
Closed as per internal conversation -- this issue will be revisited soon.

closed 2018-11-29 by shoreliner