PrintM3D/Crane #17: Crane - Quad - menu

ericl-m3d, 2018-11-29:
In the home/main screen, the numbers are set to reading the "active" temps only.

This is typically not a problem, until someone uses the "Cooldown" section.

If they use the commands in the "Cooldown" section, the nozzle temp value on the home screen will not reflect how the temp has effectively changed ("active" temp would stay at e.g. 150 C, but the tool is to "standby" temp, which is set to 0 C, but the user looking through the LCD would not see that)

shoreliner, 2018-11-29:
I am investigating the fix you have submitted for this issue.  PR will be updated and noted once issue is resolved.

shoreliner, 2018-11-30:
Had some trouble understanding the problem until I recreated it for myself.  I agree and believe your fix is suitable.  PR will be accepted momentarily; this issue will be auto-closed.

commit 8c46ecac (2018-11-30): Merge pull request #18 from PrintM3D/ericl-m3d-patch-1

Update Heat_Hotend_Off

Updated to properly reflect set temperature while cooling down on the LCD menu.

This fixes #17
closed 2018-11-30 by shoreliner