PrintM3D/Crane #23: Crane - Bowden - menu

ericl-m3d, 2018-11-29:
The knob on the LCD moves down when you rotate CounterClockwise. 

May be better to change the knob direction (rotating Clockwise to move down).

shoreliner, 2018-11-29:
Can we verify if we're now using the same LCD module for all Cranes?  Due to some past differences, it was necessary to utilize the configuration element:

`M918 R-2`

to config.g or any other of the configuration files that are called.

This previously fixed rotary direction issues on some modules,but it caused others that were working as expected to then become inverted.  We should test this with our current LCD modules and also remember to revisit testing this in firmware RC5 as it fixes many LCD issues.

ericl-m3d, 2018-11-29:
Maybe, I've only seen them on Bowdens (super early builds).
This will addressed case-by-case.

closed 2018-11-29 by ericl-m3d

ericl-m3d, 2018-12-05:
How do I reverse the knob direction? (doing a case-by-case) 

Currently, I've tried this:

```
M98 Pgeneral.g                   ; Call General Configuration Module
M98 Pnetwork.g                   ; Call Network Module
M98 Pmotion.g                    ; Call Kinematics/Motor Module
M98 Pthermal.g                   ; Call Heater/Fan/Sensing Module
M98 Ptools.g                     ; Call Toolhead Module

; Call QC start up
M98 Pqc.g

M918 E-2
```