PrintM3D/Crane #37: Crane - Quad - menu

ericl-m3d, 2018-11-30:
There is no method to pause, resume, or cancel the print, during a print session.

shoreliner, 2018-11-30:
Pause and Resume functionality are part of s_tune -- accessed via 'Tune' in the bottom right of the main menu, this only appears when the printer is in a non-idle state(Like printing, paused, etc).  Due to the state-based nature of the LCD menu,  you will only see "Pause" when printing, or "Resume" when the printer is paused as the 3rd from last item in s_tune.

Could you please verify if this is functional for yourself @ericl-m3d ?  I use this function quite often.

ericl-m3d, 2018-11-30:
This is all that appears.

[image]

[image]

shoreliner, 2018-11-30:
Noted...  I have posted the file 'MEMORYLOCATIONS.MD' here on Github, in the /SD Cards folder.  It seems that no matter which value we use here, we aren't getting a correctly returned value.  I am investigating.

shoreliner, 2018-11-30:
Investigating...

shoreliner, 2018-12-04:
@ericl-m3d -- the refresh to s_tune should have resolved this issue.  In addition, I've also collapsed the Flowrate menu into a sub menu of its own, as opposed to displaying all values actively on s_tune.  I have tested here and verified by doing the following:

1.  Start a print test file.
2.  Wait approximately 30 seconds; Tune never seems to be available directly after starting a print(it will show up, but won't be possible to select.)
3. Scroll to 'Tune'.
4. The Pause option should be available, along with a restart print option at the bottom.  Pause and Resume occupy the same space, so after activating Pause, Resume should appear.

Should this not work as expected, we should probably have a discussion related to this particular issue.  Also, in regards to step 2, it may be necessary to create a bug for this particular problem.  It's unclear to me why there should be a delay after activating a print, however, I have seen that DWC has similar delays after a print starts as well.

ericl-m3d, 2018-12-06:
This works well now. 

The cancel option is only component missing.

shoreliner, 2018-12-08:
Thank you sir -- will check out the cancel issue and update.

shoreliner, 2018-12-18:
Pull Request #60 fixes this.

closed 2018-12-18 by shoreliner