PrintM3D/Crane #40: Crane - Quad - menu

ericl-m3d, 2018-11-30:
There seems to be a syntax error in e_main menu file.
```
text R3 C5 F0 T"Fan:
alter R3 C5 F0 D0 W19 N300
```

shoreliner, 2018-11-30:
Is there something that is currently calling the e_main menu?  Two solutions exist here:  remove e_main altogether and change its references to main, or simply duplicate everything currently in main to e_main.

ericl-m3d, 2018-11-30:
I suppose e_main is called during a print session only. Not to sure.

I think it's that since the e_main content is similar to the home screen content during a print session.

ericl-m3d, 2018-11-30:
I take that back. I don't know.

shoreliner, 2018-11-30:
A lot of these files that seem to have a copy of themselves are part of a set that were utilized to present the impression of "states" before they were implemented in the Duet firmware.  At this point, it's probably best if I go through and clean up anything we aren't referencing and that isn't useful.  This one will take a bit longer and will be served with a PR instead of a direct commit so that we can discuss.

shoreliner, 2018-12-04:
Before I wrap this up:  I am going to take the current state of the menu and remove any files in /menu/ that are not actively being called.  This will exist as a patch branch initially before merging so that we can test following the removal of a number of files.

shoreliner, 2018-12-04:
As it is already a part of Pull Request #43 -- and the patch is now growing far beyond this, this issue will be closed unless it is necessary to reopen.

closed 2018-12-04 by shoreliner