PrintM3D/Crane #42: Included Gcode for Prints

shoreliner, 2018-11-30:
Can we discuss the models that have already been sliced, noted here:

http://crane.printm3d.com/community-submissions-attributions/attributions

For inclusion on the official SD cards?  I believe that we should consider making each version similar where the Quad prints with 4-color mixing, the Dual prints with 2-color mixing, and the Bowden prints solid.

Any ideas here are welcome.  It's also not time-consuming whatsoever to produce additional slices.

ericl-m3d, 2018-11-30:
I don't mind including any of these.

NOTE: There's currently a sliced model in both the bowden and quad. It's a crane micro spool holder (used for QC).

shoreliner, 2018-11-30:
Spool Holder will be included.  Fulfilled here by commit.

closed 2018-12-27 by ericl-m3d