PrintM3D/Crane #44: Unloading Macros Request

ericl-m3d, 2018-12-03:
Need an unloading macros.

(There is an unloading section, probably house it in there.)

Suggested Commands:
```
M190 S150; Needs the wait, before retraction
G1 E-100:-100:-100:-100 F600; At least 100 mm retraction, recommended speed 5 mm/s
```

shoreliner, 2018-12-04:
As noted in above commit -- remove_filament macros were added to each crane model.  Issue will be closed following addition of this to the LCD menu and any discussion necessary.

ericl-m3d, 2018-12-04:
Okay. Test them. I'll have to test them as well.

ericl-m3d, 2018-12-05:
Edits to the macros. 

```
M190 S150; Needs the wait, before retraction
M120
M83
G1 E-100:-100:-100:-100 F600; At least 100 mm retraction, recommended speed 5 mm/s
M121
```
The M120 and M121 may not be necessary. I'm following the convention of the extrusion buttons in DWC.

Definitely the M83 needs to be there.

Use the is macros. Please add to menu.

shoreliner, 2018-12-08:
Acknowledged. Will to and add to menu.

shoreliner, 2018-12-18:
Marked as RC5 milestone.  Will be resolved on merge.

shoreliner, 2018-12-18:
View the prior commit to see the updates made to x_funload -- do we need to make further changes?

ericl-m3d, 2018-12-27:
There is no unloading macros present. Please add.

Tylerun, 2019-01-03:
I am unable to attach phoot but in quad web intetface macro is present and active.

shoreliner, 2019-01-14:
Both loading and unloading macros are being overhauled for the 2.02 official release. Adding to milestone and will resolve upon branch merge of /202patch.

ericl-m3d, 2019-04-22:
Unloading options applied to in the LCD section "Loading/Unloading Filament".

closed 2019-04-22 by ericl-m3d