PrintM3D/Crane #49: Start up QC issue

ericl-m3d, 2018-12-04:
Have the menu load first, then start the QC macros.

shoreliner, 2018-12-04:
I am unsure if it's possible to force this in Firmware 2.02RC2 -- I believe a refresh option is available in RC5.  Let me do some additional research and get back to you here.

shoreliner, 2018-12-04:
Exactly. I think we should do the following for the next release:

First, after resolving the bugs here that can be resolved release version 1.32(internal 1.3, 2.02RC2) as things are now.  Then, post-upgrade, begin working on the 1.35(internal 1.3, 2.02RC5)as a separate release.  With that in mind, these issues could be rolled into a new milestone, project, or label.

ericl-m3d, 2018-12-04:
Okay.

Please finish the issues #19 and #17 . Then move exclusively to working on RC5 (or RC4, whichever you think is the easiest to work with) compatibility.

All other issues are minor enough to disregard them for awhile, until we have the "upgrade compatibility"

shoreliner, 2018-12-04:
Will do, updates coming momentarily for those in fact.

shoreliner, 2018-12-04:
As for version -- RC4 vs RC5 should only be improvements for us, so if we're making a jump we should jump to 5.

shoreliner, 2018-12-18:
Some fixes Rolled into #60 -- This will be resolved by the current open issue for a new way to load the QC menu.

closed 2018-12-18 by shoreliner