PrintM3D/Crane #57: Missing manual leveling macro in Dual and Quad.

daniel-farina, 2018-12-11:
While revising the instructions for the Crane Quad I found the following issue here: 

https://crane.printm3d.com/crane-quad-guide/manual-bed-leveling-assistant

This Quad Instructions claim there should be a calibration section called Calibration > manual bed leveling. 
[image]
but this is what I found instead: 
[image]

The first thing I did is to go to the Github repo to locate the correct Macros here: https://github.com/PrintM3D/Crane/tree/dev/SD%20Card%20Structure/Quad/macros
but it seems outdated.

Possible solution:
It seems the Calibration macros and folder is only available in the Bowden folder in Github here: 
https://github.com/PrintM3D/Crane/tree/dev/SD%20Card%20Structure/Bowden/macros/Calibration

and missing in both the Dual and Quad folder. 

Please confirm.

daniel-farina, 2018-12-11:
I suggest also giving options in the instructions: 

For example: Use the macro located at or run the following g code
```
G32;
```
We can do this using the tabs in Gitbooks. @shoreliner

shoreliner, 2018-12-12:
Issue has been confirmed and is currently in the process of being patched.  See a5188a6 for details.  The /rc5patch/ has been created to handle this update, as well as others. The current issues that have been caused by transitioning from RC2 ->RC5 are being handled separately in the rc5branch.  Following completion, it will be merged back to dev.

shoreliner, 2018-12-18:
New Leveling Macro has been added per the merge with /rc5patch/ branch.

closed 2018-12-18 by shoreliner