PrintM3D/Crane #61: Firmware Auto-update.

ericl-m3d, 2018-12-17:
For QC purposes, is it possible to write a macros or some kind of function that automatically switches firmware version of the installed duet board to a designated one (e.g. RC5)?

During QC, boards come installed with different firmware versions. Want to avoid this "checking firmware portion" and assume the QC start up will fix this.

ericl-m3d, 2018-12-27:
NOTE: Roughly tried M997 (in-application firmware update).

Could not get it to work with the DuetMaestroFirmware.bin. I think it may be filename specific, as the documentation kind of implies. (https://duet3d.dozuki.com/Wiki/Gcode#Section_M997_Perform_in_application_firmware_update)

shoreliner, 2019-01-14:
I've actually tested this successfully at this point.  I will give you an update after testing against.  This feature will be included in 2.02 Official.  For now, it's being added as a milestone release.

shoreliner, 2019-01-18:
Keep in mind you will need to ensure the DuetMaestroFirmware.bin file is located in the /sys folder.  For now, it's not possible for us to override this location.  I will keep this tagged in my notes so that we can potentially give this command more functionality in the future.

I tested this on a Crane Quad with total success -- the LCD will display a progress bar, in fact.