PrintM3D/Crane #64: QC Mode LCD Menu

ericl-m3d, 2018-12-18:
Is there a way to have a different menu option during "QC mode"?

Currently, activating "QC" mode is done by uncommenting a M98 line. 

I wanted to extend this to the LCD menu, somehow.

shoreliner, 2018-12-18:
We have some options.  I had considered perhaps making a blank menu appear on startup for 1 second or so and creating a hidden item that can be selected by pressing the rotary encoder immediately after flipping the power on switch.  I will explore some methods that will make it more of a "hidden feature".

shoreliner, 2019-01-18:
Currently investigating the viability of a splash screen in the next update.  With this, it would be possible to trigger the QC menu by pressing the button on the rotary dial while it is being displayed.

Previously, Images were not functioning as expected and simply produced garbage on the display.  With 2.02 the image rendering function has been improved and additional notes have been added to the 12864 documentation for the maestro.