PrintM3D/Crane #65: (Enhancement) Stop Print

Tylerun, 2018-12-28:
Please add a stop print command into the tuning menu.

shoreliner, 2019-01-14:
For now, I've added the Cancel Print command(M0, unconditional stop).  Do we have a reason to include stop instead of cancel?

This will be milestoned now and released in the 2.02 official patch.

Tylerun, 2019-01-15:
Nope same thing close away

shoreliner, 2019-01-18:
@Tylerun If you're available to test today, just grab /202patch branch and ensure your maestro is on 2.02 Firmware(Official Release from dc42).

I'll be fine tuning this and other options relative to how the menu system and the system configuration handle pause/stop/cancel commands.

ericl-m3d, 2019-04-22:
CANCEL option has been added. Refer to commits below:
2ba523a
9405f87
b7ceadd

NOTE: For now, we will consider STOP and CANCEL the same command.

closed 2019-04-22 by ericl-m3d