PrintM3D/Crane #66: (Bug) Restart Print

Tylerun, 2018-12-28:
Restart print does not do anything. You can press it many times and it will just keep printing.

shoreliner, 2019-01-14:
Restart and Cancel have both been updated in the latest 202patch branch.  In particular, this issue was caused by a basic typo in the menu system.  This has already been resolved in the 202patch branch; this issue will be milestoned now and closed with the merge of 2.02 official.

shoreliner, 2019-01-18:
While this is now functional I am still seeing some issues.  For example, if the M109 command is used to set hotend temp, the machine will not return the functionality to allow a Pause(printer MUST be paused in order to execute a M0).  Instead, the pause must be executed after heating.

That said, I'm doing a bit of rearrangement to the Tune(paused/printing) menu to enact a pause-only state that should be called initially.  The basic fix is already in place in /202patch/

ericl-m3d, 2019-04-22:
Restart button has been replaced with a new CANCEL option. 

2ba523a
9405f87
b7ceadd

Restart option has been temporarily removed (until further notice).
Restarting will consist of pausing, canceling, then re-selecting print file.

closed 2019-04-22 by ericl-m3d