PrintM3D/Crane #67: (Bug) Display has a value of "100" in top left corner.

bass4aj, 2019-01-02:
Top left corner of the display has a floating value of 100 over the "Nozzle" text.

Tylerun, 2019-01-03:
Which firmware and which printer are you using to get this?

bass4aj, 2019-01-03:
Crane Quad, Firmware 2.02 DWC 1.22.6, though I had this issue prior to
upgrading my firmware. I think I was on RC3.

Tylerun, 2019-01-03:
I had the same issue upon updating to release version of firmware try updating to release

bass4aj, 2019-01-03:
As in the Crane release? My firmware is on 2.02 release right now.

Tylerun, 2019-01-03:
So if you look at your printer does it say crane quad in the lower left corner

bass4aj, 2019-01-03:
I think it just says "Quad", but I will have to check when I get home.

Tylerun, 2019-01-03:
https://youtu.be/hO7-jUEo5Q4
https://youtu.be/QBarLjX7M_M

Try following one of these depending what you use. Duet interface or sd card see if it fixes it

Tylerun, 2019-01-05:
Did you get any results?

bass4aj, 2019-01-07:
Ah yeah it says crand quad

bass4aj, 2019-01-07:
Didn't have a chance to try the videos you sent. On vacation now til next
week

shoreliner, 2019-01-14:
This is typically caused by ghosting of the menu system. In other words, populating similar rows and columns as a value that already exists.  This should have been resolved in the latest official release.

shoreliner, 2019-01-14:
@bass4aj in this case you need to download the official release, which includes the /menu folder.  Because these issues were resolved with a change in the menu configuration, rather than the firmware itself, it will require you to update the entire /menu folder on your duet's sd card.

Please reopen if you have any more issues with this -- after reading over I am fairly positive this will solve your issue, however.

closed 2019-01-14 by shoreliner