PrintM3D/Crane #68: (Enhancement) Add manual bed mesh compensation through Menu

bass4aj, 2019-01-02:
The Duet can use automatic bed mesh compensation to compensate for an unlevel bed. These meshes can also be created manually. Since the encoder can now control motor movements with the latest Duet firmware, I think this would be a great user experience for bed leveling without a probe.

The flow would be something like this:
1. User selects "Mesh Compensation" from the Menu
2. Machine homes, then hovers 5mm above the front left leveling screw
3. Encoder control for the Z axis is enabled and performs 0.1 or 0.05mm movements per click.
4. User lowers nozzle until it barely touches the bed and clicks the encoder to continue.
5. nozzle moves to next screw, and repeates steps 2-4 for each corner.
6. Duet stores the height values in a bed mesh compensation (heightmap.csv) and the duet loads that file at the start of each print.

This flow also allows a software bed leveling without needed to remove a print surface from the bed

shoreliner, 2019-01-14:
Agreed -- the '202patch' branch has been created and will contain this addition. As of now, the 202patch branch contains the changes necessary to allow rotary control of the Z-offset.  This request will be milestoned and closed with the official 2.02 patch branch.

bass4aj, 2019-01-16:
Awesome. If I find any time, I will see if I can contribute at all.

shoreliner, 2019-01-18:
I've tested this and found it does produce a fairly consistent result. This is primarily because of the extremely high level of resolution provided on the Z-axis by rotary control.  Commit coming soon if anyone else would like to test this branch separately while I finish making updates.

shoreliner, 2019-01-18:
Still narrowing down the defined values, simply send a:

>M557 X15:220 Y15:220 S205

And then run a G29 or select 'Run Mesh Grid Compensation' in DWC.  I am currently 
attempting to determine the best way to deliver this to users, I believe one option to consider would be two or three macros that have growing numbers of probe points... from 4(what we have now), to 8, to 16 or something along those lines.

bass4aj, 2019-01-28:
The way I envision it working is just a guided process where the printer
will move to each point, and prompt the user to move the nozzle to the bed
with the encoder. click the encoder once its done and move to the next
point until its finished.

bass4aj, 2019-01-28:
So it looks like we can use the default case of Case 5 in bool ValueMenuItem::Adjust_AlterHelper(int clicks) to do live encoder adjustments on motor movements. Since popup menus are not implemented yet by the looks of it, we can do a Macro that appears in the menu as "Manual Bed Mesh" or something.

That macro can be pre-programmed to relaunch the same sub menu for each point that we want. The sub menu would enable the encoder to move Z and have a text telling the user to move then nozzle to the bed and click when done. When they click, the menu would return using the Return option of M32 and the primary macro would then move to the next part, executing movement gcode then relaunching the submenu.

Does this make sense? Is this feasible? or am I misunderstanding how the menu stuff works? If we launch menus from a macro, does it pause macro execution until we return from the menu, or does it just keep executing?

bass4aj, 2019-01-28:
I would start with 4 probe points and see how it goes. No need to overcomplicate at first

bass4aj, 2019-02-06:
is this code currently in one of the github branches for me to play with?

ericl-m3d, 2019-04-22:
A manual bed leveling macros (using 4 points, not referring to G32) has been applied. 

NOTE: The 4 points are the center points of the bed screws (4x).

closed 2019-04-22 by ericl-m3d